prusa3d/PrusaSlicer #10831: Improper first layer of large print

hansratzinger, 2023-06-17:
### Description of the bug

After slicing a very large print (185 x 226 x 235 mm) for Anycubic Vyper (max 245 x 245 x 260)  the first layer represents only a fraction of the entire bottom of the model.

[image]

[image]

After slicing:
[image]

The same model works in Cura fine. First layer see below
[image]

### Project file & How to reproduce

1. import model 
[KT-Muffe_DN150-20_Grad_mit_50_Schlauchanschluss_teilbar v18_Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/11778427/KT-Muffe_DN150-20_Grad_mit_50_Schlauchanschluss_teilbar.v18_Prusa.zip)

2. rotate x 90°
3. place on face
4. slice now
5. go to level 1 bottom view

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+win64

### Operating system

Win 10 Version 22H2 Build 19045.3086

### Printer model

Anycubic Vyper

mix579, 2023-06-17:
Model doesn't seem to be tilted slightly, in current position, or after rotating 90degrees. If I use F and click on the bottom surface, it rotates slightly and slices perfectly. 
[image]

hansratzinger, 2023-06-18:
Many thanks @mix579! I did not know to klick on the bottom after using F. Now ist working! Up to this model there were no need to do so. It was working only by using F.

closed 2023-06-18 by hansratzinger