prusa3d/PrusaSlicer #10835: Missing "Purge volume adjustment" in 2.6.1RC and 2.6.2rc

Mick-kohn, 2023-06-17:
### Description of the bug

Missing "Purge volume adjustment" in 2.6.1RC and 2.6.2rc

### Project file & How to reproduce

no files

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.x

### Operating system

windows

### Printer model

none

Stefan-AUT, 2023-09-19:
2.6.1+ Win64 on Windows 11
I dont have a "Purge Volume..." Button in the Slicer!?

[image]

Version 2.6.1+win64

lukasmatena, 2023-09-20:
Intended behavior. From [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6) change log:

> The 'Purging volumes' settings accessible by a button in the right panel is now only shown and used for single-extruder multi material printers. Setting purging volumes based on which filament is unloaded does not make sense for proper multi-extruder systems, which can easily control the required purging in Filament Settings->Advanced->Minimal purge on the wipe tower. Note that this is potentially a breaking change, affecting non-single MM profiles with the 'Purging volumes' set much higher than the value in filament settings, in which case the wipe tower will become smaller than it was in previous PrusaSlicer versions.

closed 2023-09-20 by lukasmatena

1616h, 2023-11-10:
"only shown and used for single-extruder multi material printers." - I've loaded up a Geeetech A20T & A30T and don't have this button. Need to change purge amount as going from black to white with just the automatic wipe tower is not enough.