prusa3d/PrusaSlicer #10836: Executing an external program before uploading print

stefanct, 2023-06-18:
Since it's now easily possible to directly start prints on remote network printers from within PrusaSlicer it would be nice to be able to turn on the printer automatically too before starting the print. To that end, it would suffice (for me) to be able to execute a script synchronously before PS uploads. The script would then turn on a remotely controllable power plug and wait for the printer's network interface to become ready before returning control to PS.
Other use cases could for example be enabling exhaust fans, archiving *.stl files, or starting a video recording of the print.

foreachthing, 2023-06-19:
This might help: https://help.prusa3d.com/article/post-processing-scripts_283913

stefanct, 2023-06-19:
The post processing scripts are executed at slicing time and/or g-code export time (the documentation on the referenced page unfortunately does not say anything about the execution timing/scheduling). I don't see how this "might help" executing instructions at print time as they might not be the same. Assuming they are executed shortly before the gcode is sent to the printer, is it possible to distinguish this invocation within the scripts from other executions where gcode is produced? Does it also happens on reprints or is the gcode stored/cached somewhere and the scripts only executed once if their input does not change?

bubnikv, 2023-06-19:
Post-processing scripts | Prusa Knowledge Base (prusa3d.com)
<https://help.prusa3d.com/article/post-processing-scripts_283913>
 SLIC3R_PP_HOST
is set to the name of the host you are to send your G-code to
PrusaLink, Octoprint ...

stefanct, 2023-06-20:
Yes, that seems to work. PS waits with the upload till the script is done (naturally, because the script might change the data sent to the printer in the expected use case). The documentation could be a bit more clear (e.g., when exactly the scripts get executed, if they are run serially) but overall I was just not looking at the right stuff.
One thing that could definitely be improved is the user feedback: there is no indication that the script is run. It would be great if there would be *some* kind of activity indicator (ideally the script could report back the progress/liveliness via a pipe or similar).

For reference, this is what I use to turn the Sonoff Tasmota attached to the printer on and wait till its webserver becomes responsive:

    #!/bin/sh

    printer_host="prusa_mini"
    plug_host="sonoff"
    plug_user="admin"
    plug_pwd="12345"

    if [ "${SLIC3R_PP_HOST}" != "PrusaLink" ]; then
      exit 0
    fi

    while [ "$(curl --max-time 5 --silent --write-out "%{response_code}" -o /dev/null "$printer_host")" != "200" ]; do
      curl -s -m 5 -X POST "http://${plug_host}/cm?user=${plug_user}&password=${plug_pwd}&cmnd=Power%20on"
      sleep 1
    done

Feel free to close this if you think the points above are not worth further discussion. Thanks for the help!

foreachthing, 2023-06-20:
You could just display a progress [message | bar] in your script.
I've done so with python and that works. PS will call the script, your script shows a [command prompt | terminal window] with your message.

stefanct, 2024-06-01:
Hm, after an upgrade (to 2.7.4) the above script no longer works. The slicer prints an error message containing "PrusaLink: Error getting storage: Couldn't connect to server" before the script is even executed AFAICT. This is the result from calling `get_storage()` at https://github.com/prusa3d/PrusaSlicer/blob/dba0377c298ab2752d2f80b5ca89d619bb984415/src/slic3r/GUI/Plater.cpp#L7399
I am not 100% sure about the order of execution. Is this supposed to be executed before post-processing?

closed 2026-09-29 by prusaslicer-bot[bot]