prusa3d/PrusaSlicer #10837: G92 E0 not needed

landodragon141, 2023-06-18:
### Description of the bug

Should G92 E0 be inserted automatically in G Code when relative E is not enabled?

### Project file & How to reproduce

Disable Relative E. Slice project and inspect Gcode

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.6.0 RC2

### Operating system

Windows 10

### Printer model

Custom printer with Marlin 2

bubnikv, 2023-06-19:
Why do you think G92 E0 not needed? We have an experience with errors
accumulating in a floating point accumulator, leading to weird extrusion
effects.

landodragon141, 2023-06-19:
Sorry it was late and I obviously had a bit of a brain fart.

closed 2023-06-19 by landodragon141