prusa3d/PrusaSlicer #10839: Centering in object coordinates broken in 2.6.0-rc

nordurljosahvida, 2023-06-18:
### Description of the bug

In the new object coordinates interface introduced in 2.6.0 there is a bug that has been around since alpha where often you cannot  center a part to zero in object coordinates mode. when you enter 0, a different value is stored and visualized, in a way that you can only reduce the value but never get it to zero. Also, "place on plate" button doesn't work in object coordinates. This is happening all the way in rc2 and IMO is a release blocker but of course I have no idea.
Video 1: https://drive.google.com/file/d/16V31xJeGg1KKKoDOfDOmh3L8bGx1aKx7/view?usp=sharing
Video 2: https://drive.google.com/file/d/16Y5Bqjya-PAuJ5LzD_Ohw51jYkiEw75O/view?usp=sharing

### Project file & How to reproduce

open the 3mf in rc2
delete the small cylinder.
right click on the big remaining cylinder and "add part" -> cylinder.
ensure you have object coordinates positioning selected in bottom right of the screen [the idea is you're trying to center the new cylinder on the old one]
make the new cylinder 8mm in diameter just to make it smaller than the first one.
enter 0 for both X and Y coordinates.
coordinates will not save as 0 but will change to a seemingly random other number
[spacers.zip](https://github.com/prusa3d/PrusaSlicer/files/11781493/spacers.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-rc2

### Operating system

win11

### Printer model

mk3s+

Jan-Soustruznik, 2023-07-10:
SPE-1789

glennimoss, 2023-08-21:
Just ran into this bug myself.

lukasmatena, 2023-09-05:
Fixed in [2.6.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1). Closing.

closed 2023-09-05 by lukasmatena

nordurljosahvida, 2023-09-17:
great work thanks!