prusa3d/PrusaSlicer #10842: 2.6.0 RC2 (Filament runout not detected)

darrellxls, 2023-06-18:
### Description of the bug

PrusaSlicer 2.6.0 RC2 filament runout issue. I have discovered between RC1 and RC2 something has changed, but I can't figure out what. any file sliced with RC2 if I start with a little bit of filament it seems that the MK4 doesn't detect a filament runout, and will keep printing with no filament coming out.

I have stepped back to 2.6.0 RC1 and did the same test and it will detect a runout and allow me to unload and reload the next filament. 

### Project file & How to reproduce

how to reproduce? just slice anything with 2.6.0 RC2 and have a small piece of filament loaded so that it will runout.
[Shape-Box(filament runout test).3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11782426/Shape-Box.filament.runout.test.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-rc2+MacOS-arm64

### Operating system

MacOS Ventura 13.1

### Printer model

MK4IS alpha

darrellxls, 2023-06-21:
> Hello, @darrellxls ,
> thank you for your feedback, but this seems like problem of the Firmware or Hardware - Filament sensor. Could you contact the support, please.
> I suggest to close this issue.
> thank you, have a nice day

Did you try to replicate it?Sent from my iPhone

GuidoHendriks, 2023-06-21:
When slicing your 3mf file in RC1 and the release version, the only difference is the slicer version and time on the first line in a comment, that's not going to break your filament runout. Better reach out to support, they'll take care of you. 👍🏻

closed 2026-09-29 by prusaslicer-bot[bot]