prusa3d/PrusaSlicer #10849: Disable bridge completely

Neo2003, 2023-06-19:
**Is your feature request related to a problem? Please describe.**
When doing multiple colors print by changing filament and run a second print, I am unable to disable bridge even if I DO KNOW there is supporting material.

**Describe the solution you'd like**
A check box to disable bridging generation all together.

**Describe how it would work**
See this stupid example
This is layer 1, all the holes are already filled by a previous print (0.20 mm high)
[image]

But no way, it will bridge on next layer, so I have to make 2 or 3 layers on top to hide the poor quality of the bridging.
[image]

**Describe alternatives you've considered**
No other idea than a checkbox to disable bridging. Maybe you can point me to the code where I can disable bridging, so that I can build a separate version for specific purpose.

**Additional context**
This would also be useful when we use support, bridging on top of the support has always been a bad choice.

Jan-Soustruznik, 2023-06-21:
Hello, @Neo2003 ,

thank you for your feedback. I hope that I correctly understood .. please help me. You printing twice  .. first one layer with different color(filament) and then second for the base with different color(filament). So you fill the empty space in the base with the first print? - right?

matveimug, 2023-11-14:
Having the same problem with the same usecase - yes, previous print is with a different color/filament, no need for bridging on top of that.

CaptainFalcon92, 2023-11-20:
Hello bumping up this very simple yet required feature. 

I would need to **turn off bridging entirely** too.
I want slicer to just **print like there was no void underneath**. I insert physical metallic support element; there will be support, though it's not the slicer's concern. Can slicer do that please ?

TheWortLord, 2024-04-21:
I would love this feature as well. I do two color prints quite often using manual filament change and the single extruder multi material option. Would love for my second layer to print like a normal second layer.

carlosjln, 2024-04-28:
Sighs... I'm trying to optimize this print for volume production and I really don't need a bridge fill over a 0.6mm diameter hole.

[image]

Neo2003, 2024-09-16:
> Hello, @Neo2003 ,
>
> thank you for your feedback. I hope that I correctly understood .. please help me. You printing twice .. first one layer with different color(filament) and then second for the [...]

You are right, I print twice, so the second print is lying on the first one which is only 1 layer high. There is simply no possibility to tell the slicer to ignore any bridge or support.

unixb0y, 2026-09-02:
Someone please check whether the issue is still present. I highly doubt that 3.0.0 removes the bug though, unless specifically mentioned.