prusa3d/PrusaSlicer #10852: Replace (with STL) not available when using STEP models

ulope, 2023-06-19:
**Is your feature request related to a problem? Please describe.**
When working with STL models the very convenient "Replace with STL" function can be used to load a new model while keeping all already made modifications in place.

This option is greyed out when the model loaded came from a STEP file (which I guess technically makes sense since I don't want to replace it with a STL). I do want to replace it with another STEP file though.

**Describe the solution you'd like**
When a model was loaded from a STEP file the current menu item `Replace with STL` should be replaced (ha ha) by an menu item `Replace with STEP`.

**Describe how it would work**
Same as with STL files.

**Describe alternatives you've considered**
I can't think of a good alternative.

foreachthing, 2023-06-19:
No, if you want to _replace_ it with an STL, then it's correct. But, "Replace Model with ..." would make more sense, since you can choose multiple sources (stl, step, 3mf, ojb, ...).
That's nitpicking, of course 😉.

ulope, 2023-06-19:
Actually I was being confused by the UI. When a model has modifiers attached you have to right click on the actual model node not on the top level root. This confused me (and ahould probably be changed...).

But apart from a labeling issue there is really nothing not working here.

ChrisHeerschap, 2023-12-11:
@ulope was trying to figure out why I couldn't replace a part with another and your answer was exactly what I needed - thank you!

closed 2026-09-29 by prusaslicer-bot[bot]