prusa3d/PrusaSlicer #10855: Arachne generator producing gcode with too rapid E speed changes

mbbill, 2023-06-19:
### Description of the bug

While printing some simple circles with parameter value set to very high which essentially printing solid layers, I noticed my printer stalls rapidly printing the first two inner perimeters. Then I checked the gcode and noticed some g-code like below:
```
G1 X100.741 Y82.703 E.03422
G1 X100.812 Y82.672 E.00204
G1 X102     Y82.166 E.0342
```
The extruder speed changes over 10x during the three lines, and with the ejerk settings, the printer has to slowdown to accommodate the rapid change. Turning up the EJERK helps a little bit but I don't think most printers can really catch up with this speed.

The `classic` generator works well.

[image]

### Project file & How to reproduce

[FanOutletRing_0.2mm_PLA_ENDER3_16m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/11793913/FanOutletRing_0.2mm_PLA_ENDER3_16m.gcode.zip)
[FanOutletRing.zip](https://github.com/prusa3d/PrusaSlicer/files/11793915/FanOutletRing.zip)

To reproduce the issue just set the perimeters to a very high value, like 10.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2 release

### Operating system

Windows 10

### Printer model

Ender3Pro

StephS, 2023-06-25:
While it looks like the extruder speed is changing by a factor of 10, you need to remember that the extrusion is linear over the length of the move.
So to find the actual extruder speed, we need to calculate the length of the move, which is the distance from the previous extruder position.
`5258: G1 X99.572 Y83.255 E.03424
5259: G1 X100.741 Y82.703 E.03422
5260: G1 X100.812 Y82.672 E.00204
5261: G1 X102     Y82.166 E.0342`

With these 4 lines, we can calculate 3 distances using the following formula d = √((x2 - x1)^2 + (y2 - y1)^2)
[5258-5259] = 1.292347
[5259-5260] = 0.0774726
[5260-5261] = 1.29127

We can divide these lengths by the extrusion length to get an extrusion constant ratio, how many mm of extrusion:
1.292347 / E.03422 = 37.76583
0.0774726 / E.00204 = 37.97676
1.29127 / E0.0342 = 37.75643
That's about 37.8mm of length of travel per 1mm of filament (or 1/37.8 mm of filament per 1mm of travel). This ratio should be constant for all extrusions of the same width.
Your extruder speed is directly related to this ratio.
So looking at this, your extruder and movement should be going nearly the same speed for these 3 moves.

However, this doesn't take into account your controller board speed and firmware. If you have an 8-bit control board, this 0.077mm distance move might be too short. Your controller needs to calculate acceleration/deccel, junction deviation speed (or jerk), and other functions within the move. 8-bit controllers can struggle with this. Even 32-bit can if the moves are too small or too fast.

But what you are seeing is also an artifact of how the Arachne works. The Arachne perimeter generator actually ignores the g-code resolution setting and instead uses hard-coded values ( see https://github.com/prusa3d/PrusaSlicer/issues/10593 )
This means that Arachne generator makes much higher resolution output than the classic - and as of right now you can't change it.

At any rate the issue is:
1: Arachne is too high of resolution and can't be changed,
2: Your controller board/firmware can't keep up with the rate of data that the Arachne generator creates.

mbbill, 2023-06-25:
Thanks for the explanation! I'm using marlin with a 32bit arm board (STM32F103). I'll do more test and post the updates here.

mbbill, 2023-07-01:
I think it might be related to the printer's firmware, closing the ticket.

closed 2023-07-01 by mbbill