prusa3d/PrusaSlicer #10856: Import Config Bundle Assertion Error

clstr, 2023-06-19:
### Description of the bug

## System Details
Arch Linux (Manjaro)
Lenovo ThinkPad X1C 5th W10DG
Intel® Core™ i7-6600U
16.0 GiB RAM
Gnome 43.5 (Wayland)

## Description
When attempting to import a config bundle I get an error.

```
An Assertion Fail

/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

ASSERT INFO:
/usr/src/debug/prusa-slicer/deps/destdir/usr/local/include/wx-3.1/wx/strvararg.h(480): assert "(argtype & (wxFormatStringSpecifier<T>::value)) == argtype" failed in wxArgNormalizer(): format specifier doesn't match argument type

BACKTRACE:
[1] g_closure_invoke
[2] g_signal_emit_valist
[3] g_signal_emit
[4] gtk_widget_activate
[5] gtk_menu_shell_activate_item
[6] g_signal_emit_valist
[7] g_signal_emit
[8] gtk_main_do_event
[9] g_main_context_dispatch
[10] g_main_loop_run
[11] gtk_main
[12] __libc_start_main
```

### Project file & How to reproduce

https://gist.github.com/clstr/f4b7a6025400913b9daea11c252f82c2

On Windows
Create custom Printer/Filament/Printer presets
File > Export Config Bundle...

On Target Machine
File > Import Config Bundle...

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2-1

### Operating system

Arch Linux (Manjaro)

### Printer model

N/A

Jan-Soustruznik, 2023-06-21:
Hello, @clstr , 

thank you or your feedback, but we cannot reproduce reported problem. I tried on Ubuntu 20.04.6 LTS and import passed without problem. 

Could you check if you have all necessary permission or if problem may be caused some OS settings?

Thank you, have a nice day

lukasmatena, 2023-06-21:
@clstr What is build "2.5.2-1"? Does the same issue happen when you run an AppImage from your github?

clstr, 2023-06-21:
@lukasmatena I will try it, that version was in the repo for my distro.

@Jan-Soustruznik I will check.

closed 2026-09-29 by prusaslicer-bot[bot]