prusa3d/PrusaSlicer #10858: Organic support not working properly

Excel265, 2023-06-20:
### Description of the bug

When trying to generate organic support on a model tilted at 45 degrees (simple cylinder) 
It gets frozen at 70% and then the slicer crushes

### Project file & How to reproduce

1. Add Shape  -> Cylinder
2. Change Dimension to : Diameter 10 mm, Height : 20 mm
3. Tilt 45 degrees 
4. Generate Oragnic supports 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 rc2 

### Operating system

Win10

### Printer model

not relevant

Jan-Soustruznik, 2023-06-20:
Hello, @Excel265 , 

could you share project file (3mf) where problem occur? I cannot reproduce issue according to steps above.

thank you, have a nice day

Excel265, 2023-06-20:
[image: image.png]

forgot to enable the supports, sending a new file

Jan-Soustruznik, 2023-06-22:
@Excel265 , I cannot find any file attached for this ticket, Could you try to send it again, please?

Excel265, 2023-06-22:
Hi, I sent it via email didn't find an option to upload it to in the github
interface

foreachthing, 2023-06-22:
@Excel265: you just drag&drop a file in the "write" box, when commenting.
Or, you click on the "button":
[image]
just below the "write" box.

I could not recreate your issue in 2.6.0
[image]

Excel265, 2023-06-22:
The attched files

foreachthing, 2023-06-23:
Here's a similar issue: https://github.com/prusa3d/PrusaSlicer/issues/10762

And I've found the issue here: **please disable RAFT**. 
I disabled the raft in your 3mf and it sliced fine. I did not create any support but that's another issue. After painting on some support it sliced and got support (I'm rambling ...). Anyways, as soon as I set Raft Layers to 1, it stopped slicing at 70% and crashed.

After another tyr, PS crashed after Raft Layer 4. ... so, something is going on with rafts.

PS talks about raft here: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-beta2
But something's still off.

Excel265, 2023-06-23:
> Here's a similar issue: #10762
>
> And I've found the issue here: **please disable RAFT**. I disabled the raft in your 3mf and it sliced fine. I did not create any support but that's another issue. [...]

Yep, I was able to generate supports when raft is disabled it. 
Weird thing is, when setting the raft layers to 2 and 3 it sliced it, and as you said for 4 layers of raft it crashes again at 70%

bubnikv, 2023-08-29:
The issue was indeed connected to raft. I twill be fixed in PrusaSlicer 2.6.2 final.

closed 2023-08-29 by bubnikv