prusa3d/PrusaSlicer #10864: Negative Volume 'hole' is not exporting as STL

WhiteDragonD, 2023-06-20:
### Description of the bug

I loaded a STL, added two negative boxes for an L shaped hole, exported, the file had the hole in it.

I then imported that file, added another box, a negative hole, instance... exported and the new file did not have the new negative volume hole that was added.  

### Project file & How to reproduce

Not sure if this is exact steps but... like the description, import file, add solid box part, add a negative cylinder, rotate, move, add a negative box around solid box, add instance, export.  The export file will not have the hole.

[SprayCanShelfBracket.zip](https://github.com/prusa3d/PrusaSlicer/files/11803090/SprayCanShelfBracket.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 RC2

### Operating system

Windows 10

### Printer model

Prusa i3 mk2.5s

sarusani, 2023-06-20:
`The algorithm may fail on non-manifold geometries (e.g., those containing self-intersection, etc). In that case, the user is notified, and only positive volumes are exported`

If you want have an STL with the hole you need to use a proper CAD editing tool. But you could share the 3MF, it will contain the negative volume.

WhiteDragonD, 2023-06-23:
How do I fix it? Can meshmixer do something? can I convert it with another app and save it somehow so I can import and add negative then export w/ the holes?

I don't understand and I was hoping the new "export with negative" feature would work.

thanks

WhiteDragonD, 2023-06-23:
Ok I have another comment to add..... I've been doing some testing and how I reproduced this is...

Import a STL.
Add Part (box) to object.
Export as new STL.
Drag new STL into PrusaSlicer
Add Negative Cylinder
Export as New STL.
Drag that new STL into Slicer.
Add another Negative Cylinder.

This will fail on exporting the 2nd negative piece.

AndrewFarley, 2024-11-17:
Can some active contributor of this codebase please explain why this isn't possible or isn't implemented?  It is just a "someone hasn't spent the time/energy to do this?" or is this a deeper technical issue that would require significant refactoring of some core/internal logic that makes it a huge barrier to implementing this.

I'd really, really like this feature because I often find that switching back to a 3d CAD program then re-exporting is quite fiddly and takes so much more time than if I do certain things (such as text, or making a hole for something) is faster in the Slicer, but then I can't export my results so it kinda makes me spend/waste more time doing my edits always externally, exporting the file, importing into the slicer etc.

Can we please get someone familiar with the code to comment with some background/history/technical details?  I ask because I would consider implementing this myself, or paying a consultant to implement this, as this would greatly help me.

Thanks

AndrewFarley, 2026-09-05:
I'd still really like this.  Last minute modifications to a model via negative objects is really useful but then you're stuck and can't export them to save them and/or to edit them in another program, they're always stuck in your slicer.  So then you have to re-create in another program the negative object you already did in the slicer.  Please keep