prusa3d/PrusaSlicer #10865: Connectors - allow editing after creation

DFEvans, 2023-06-20:
**Is your feature request related to a problem? Please describe.**
When experimenting with connectors, a fair bit of trial and error is needed to configure sizes and tolerances. However, there does not appear to be a way to edit existing connectors - the only option is to go back in the undo history to the relevant Cut. If the user has done many operations that they don't want to undo, this results in a loss of work.

Additionally, going back to the previous Cut is error prone, as changing connector settings after going back does not apply them to placed connectors *unless* the user selects the connector first - even though the user's previous input values were remembered.

**Describe the solution you'd like**
Add some method to edit connector sizes (and possibly placement) later.

**Describe how it would work**
Allow the user to bring up the connector settings UI again for a pair of connectors.

**Describe alternatives you've considered**
When going back in the Undo history to a Cut, and opening the connector settings menu, changing the settings applies them immediately to existing connectors. However, I think PrusaSlicer allows multiple connectors with different properties in the same Cut, and it would be challenging to decide which connector to select in an intuitive way.

**Additional context**

Cutieguwu, 2023-07-07:
This is much needed in projects that are sliced to be completed quickly. I've printed a proof of a model which I didn't add adequate tolerances to my connectors for. This print in particular uses a 0.6mm nozzle and 0.3mm layers to cut down on time.

A feature like this would help to quickly resolve user caused post-printing problems due to inadequate tolerances, poorly chosen connector style for the use case in the project, poor placement of the connector for the use case in the project and just allow for an easier experience while learning how to use the cut tool's features as you need not constantly undo and reselect the model and cut tool (which thankfully does remember the position of the slicing plane, making it at least a little bit easier to tinker with before slicing and printing.)

Icy-Ion, 2023-08-21:
I agree that it would be extremely helpful to be able to edit the connectors feature after performing a cut. This would allow considerably smoother workflow should your placement not be optimal, or after an initial print coming to the realization your connector tolerance was not sufficient for a good fit.  
We are grateful for time and effort you out into all of these requests and fixes. Please keep up the fantastic work!
Thank you,
icy

nrnrnr, 2023-10-07:
This would be a great feature!  (I'm a newbie, trying to reconstruct the tolerances on snap connectors that worked well.)

gaitskell, 2023-11-24:
I also really support this. It is tough iterating on optimal connectors when you have to recreate everything each time. Please add editing existing connectors if at all possible. Many thanks,

zabbidou, 2023-12-13:
I just printed a model and the connectors were too flimsy. I wanted to change them to be bigger, but my parts have paint on seam and supports that I just dread doing again, just to change a connector.

gegy1, 2024-01-22:
Yes, very neccessary feature to edit cut and connectors after finished the cut

3dPrinting1127, 2024-05-15:
I had to increase my extrusion multiplier and now all my connectors don't fit into one another.  At the very least they should allow us to increase/decrease the size of the connectors while maintaining their position.  I print some pretty complex parts that require lots of paint on supports, modifiers, etc.  This is insane that I have to do it all over again.

Icy-Ion, 2026-09-03:
Keep open