prusa3d/PrusaSlicer #10869: View orientation box/cube

mlisner, 2023-06-21:
It would be nice and useful to use a 3D cube or box symbol for the 3D view orientation, which could be visible in the one corner of graphic area, e.g. top-right, Not only the keys on numeric keyboard.
 
 
[image]

mlisner, 2023-06-21:
Additional information - when the view is perpendicular to the axis, it is possible to turn to another perpendicular view in CAD programs by clicking on the arrows (which appear).
Just inspiration from other programs.
[image]

kubispe1, 2023-06-21:
Thanks a lot. It is a duplicate of #4410
But I think it is quite useful especially for cad oriented users

closed 2023-06-21 by kubispe1