prusa3d/PrusaSlicer #10877: -z and homing check for prusa slicer

3DManuMa, 2023-06-22:
For prusa slicer I would like to have a very simple feature which could saves a bunch of filament and print life's:

 

Some times, had that even with my mini, the printers decides to kiss the part after successful printing and ruins the whole print which is very annoying. 

 

Prusa slicer got many warning massages in the last time which is quite nice to have.

 

So, my wish: why not implement another warning massag which Checks if there's a honing or at least a -Z moving command in the end of the gcode? This would be easy to do and also great to have. (Maybe also on the viewer?)

rtyr, 2023-06-22:
> Some times, had that even with my mini, the printers decides to kiss the part after successful printing and ruins the whole print which is very annoying.

It sounds like very old firmware issue. Are you sure you are using up-to-date firmware on your MINI?

3DManuMa, 2023-06-22:
After some updates I didn't had this issue not anymore but I was printing a lamp on my CR10 and sliced with prusa slicer 2.5 or even 2.6 and had that issue today in the morning. But I don't know how actually the Firmware of the cr10v3 is

rtyr, 2023-06-22:
I don't think this is a task for PrusaSlicer. We cannot prevent all existing scenarios caused by incorrectly configured profiles or firmware issues. For example, your CR10 can disable Z axis steppers after the print, which can result in some drop down.

closed 2026-09-29 by prusaslicer-bot[bot]