prusa3d/PrusaSlicer #10879: First layer extrusion width

tom-321, 2023-06-22:
### Description of the bug

I'm trying to print a simple part with 1mm nozzle
There no fine details so it would be much faster with the 1mm nozzle

But I wonder about the given default values, 'cause the first layer has 200% referring the layer height, which means it has 0.8mm if I change layer height from 0.2 to 0.4 -> that's still less than the nozzle diameter ?!
If I am correct then this does not work.

### Project file & How to reproduce

[Untitled.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11833636/Untitled.3mf.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+MacOS-x64

### Operating system

macOS Ventura Version 13.1 (22C65)

### Printer model

ender3

rtyr, 2023-06-22:
First layer extrusion width is calculated over first layer height if expressed as a percentage. This behavior inherited from upstream Slic3r is sort of counter-intuitive and may be changed in the future (backward-compatibility is the main issue here). 

Setting absolute values will be better solution for you.

tom-321, 2023-06-22:
I set it to 0 so it's automatically calculated and it's working fine.

The thing I considered was the given default value, which as I see it does not make sense.

treaves, 2023-06-23:
> First layer extrusion width is calculated over first layer height if expressed as a percentage. This behavior inherited from upstream Slic3r is sort of counter-intuitive and may be changed in the future (backward-compatibility is the main issue here).
> 
I'm trying to understand the response; backward-compatibility with what?  It can't be Slic3r, as that's very dead. So are you referring to prior versions of this product?

rtyr, 2023-06-23:
Yes. It would break all existing profiles, 3mf projects, etc. which are using percentage. But we are still planning to change it.

closed 2023-06-23 by rtyr