prusa3d/PrusaSlicer #10881: After slicing and then changing settings usually re-slice is not an option only Export G-code

buccaneer-jak, 2023-06-22:
### Description of the bug

After slicing and then changing settings, usually re-slice is not an option only Export G-code. Only by closing the program can I re-slice. I later edited a .ini file and I had the option to re-slice for 5-6 tries and then it stopped again.

Re-slice should be a permanent option

### Project file & How to reproduce

[Test_model Pruser Test 1.zip](https://github.com/prusa3d/PrusaSlicer/files/11841127/Test_model.Pruser.Test.1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

Fokoos Odin-5 F3 but also tried with Creality Ender 3 profile

buccaneer-jak, 2023-06-23:
Update. Eventually even closing the program stopped working. The option is locked at Export G-code no matter trying to load a new project or importing.
I uninstalled 2.6.0 and tried 2.5.2 but no option to Slice, still locked. I have never previously installed Prusa so perhaps I should have started with an older version. About to reboot and see what happens.

lukasmatena, 2023-06-23:
Could you please describe exactly what you changed? A short video would be even more helpful. Thanks.

foreachthing, 2023-06-23:
@buccaneer-jak, please check your preferences and confirm "Background processing" is _disabled_.
If it's enabled, it might re-slice quickly and PS lets you know you can "Export G-Code"! Not a bug, feature! 😉

buccaneer-jak, 2023-06-23:
I just tried to make a video with (windows key) Alt R  but it cuts off the bottom of the video show the Export G-code box part. The editing I did was based on this video, deleting some lines in the .ini but slicer worked after the edit.
I was just using the test cube from Fokoos. Uninstall didn't work I also had to delete the Prusa directories where the .ini resided plus other settings

I again had the problem but the Creality Ender 3 printer  seems to work correctly now

This .ini came from a FB user group as I have waited 6 days to join the Official group but they Fokoos seem to be asleep at the wheel.

[Fokoos Odin-5 F3 Prusa Slicer Config PLA.zip](https://github.com/prusa3d/PrusaSlicer/files/11845766/Fokoos.Odin-5.F3.Prusa.Slicer.Config.PLA.zip)

buccaneer-jak, 2023-06-23:
foreachreaching I don't think that's it as I tried both methods of "Background processing" . Also at one stage I was just changing the supports and infill options and for 5-6 changes I could re-slice and then the option suddenly stopped.

earldbjr, 2023-06-24:
I've had the button appear stuck before, can you switch modes/reslice using the TAB key?

rcmaniac25, 2023-06-25:
I've had this happen a number of times. I use background processing and can confirm no progress information is shown, but eventually the screen updates to show the finished slicing info.

buccaneer-jak, 2023-06-25:
> I've had this happen a number of times. I use background processing and can confirm no progress information is shown, but eventually the screen updates to show the finished slicing info.

And does it let your re-slice if after make changes? Anyway pretty sure that's not it as I tried both options. And one time I  kept making one change and it let me re-slice 5-6 times and then the option didn't come back. I'll try again this week.

rcmaniac25, 2023-06-25:
If the slicer had 3 states: before changes, actively slicing and showing progress, and done slicing and showing the sliced results... the middle state gets skipped and the slicer appears to be stuck/frozen. With all changes seemingly not doing anything.

But if I wait long enough, it eventually jumps to the last state (done slicing...) and any changes I do then will run like normal, often showing progress and such.

Why it ends up without showing progress at times, I don't know. It's not frequent, but frequent enough I've played around to see if it comes back and what might trigger it (I never found an actual trigger).

buccaneer-jak, 2023-06-26:
I have found the .ini file from Warner Soditus on the Official Fokoos FB group does seem to work so hopefully this info can be useful for someone else looking for the correct .ini file.
The problem as he describes may be because the Official file has 'print' and 'filament' settings mixed in.

https://www.facebook.com/groups/fokoos/permalink/5085138514836883

I'm going to close this based on the above possible cause of the problem.

closed 2023-06-26 by buccaneer-jak