prusa3d/PrusaSlicer #10882: Missing parts of first layer

neclimdul, 2023-06-23:
### Description of the bug

For some reason whenever I slice the pi4b case from BTT, the lettering doesn't completely connect on the first later. It doesn't seem to be a problem with the STL as the part that is missing is sometimes different. Additionally, lowering the print below the print bed still leaves the missing section.

For what ever reasons this seems to be specific to PrusaSlicer. Cura and Orca sliced the lettering fine.

Prusa 2.5.2 and 2.6.0
[image]
Orca
[image]

### Project file & How to reproduce

https://www.printables.com/model/269871-btt-pi4b-adapter-case

[PI4B Top.zip](https://github.com/prusa3d/PrusaSlicer/files/11842084/PI4B.Top.zip)

Steps used:
 - Loaded stl from printables
 - flipped to face.
 - slice.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+linux-x64-GTK3

### Operating system

Ubuntu 22.04.2

### Printer model

Ender 5 Pro

mix579, 2023-06-23:
Set elephant foot compensation to 0 and it looks okay in the preview. I suspect the missing letters are a bit thinner or not in the same plane as the others.

neclimdul, 2023-06-23:
Oh! Great suggestion! I'll double check this evening but that makes perfect sense.

neclimdul, 2023-06-23:
Yeah, that was it.

The plane of the letters doesn't _seem_ like its connected because as noted I can lower below the build plate where the build plate should be the plane it doesn't change. Also the letters look to be blocks so hard to believe their thinner but maybe? 

It would be cool if one of those new warnings was triggered when something like this happened or it didn't happen at all but I'm happy chocking this up to user error and a lesson learned. Thanks!

closed 2023-06-23 by neclimdul