prusa3d/PrusaSlicer #10885: Not at all monotonic pattern at the bottom and top layer (except the last layer)?

juckendesAuge, 2023-06-23:
### Description of the bug

Why does Prusa Slicer since the first 2.6.0 alpha not always print alternately the monotonic pattern at the bottom and top layer (except the last layer)?
Is it the option "Improved Ensure vertical shell thickness" (See the picture)?

What I'm still interested in, does the stability suffer?

My Version is  "PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220.AppImage" runs on Fedora 38

[image]

Thanks a lot

### Project file & How to reproduce

Open a stl and slice it.
Check the bottom layers.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220.AppImage

### Operating system

Fedora 38 

### Printer model

Prusa MK3s+

neophyl, 2023-06-23:
It has never done that.  Only the very bottom and the very top layers used the fill patterns and the rest used rectilinear.
With the new algorithm for solid infill that is being used in 2.6 you now get the almost concentric like patterns instead of the rectilinear.

Btw this issue is a duplicate https://github.com/prusa3d/PrusaSlicer/issues/10573

juckendesAuge, 2023-06-23:
Thank you very much.
I have a hard time searching with dublikat. I do not know how it was named.

closed 2023-06-23 by rtyr