prusa3d/PrusaSlicer #10886: Allow modifiers to overwrite retraction settings

okelk, 2023-06-23:
**Is your feature request related to a problem? Please describe.**
There's no way in PrusaSlicer to change retraction settings inside one print (at least without hacky ways of doing it like setting up multiple extruders). It would be great to be able to do this - mostly for tuning ( make a retraction tower for instance ). but also for niche cases of print time optimizations (small features only on part of object.)

 I want to be a able to change retraction settings with modifiers (blocks or height range) or in the "+" menu for color changes/custom gcode

rtyr, 2023-06-23:
There are already issues on that topic, for example https://github.com/prusa3d/PrusaSlicer/issues/3635. I am closing this as a duplicate.

closed 2023-06-23 by rtyr

Kekht, 2023-10-02:
I think topicstarter meant not for height range, but for modifier volume (so to make different retracts inside each layer)