prusa3d/PrusaSlicer #10889: PrusaSlicer 2.6.0 AppImage and binary do not launch without system Wayland libraries

KyleMaas, 2023-06-23:
### Description of the bug

I will note that I am on Gentoo Linux, up-to-date but without Wayland support.

Downloaded the 2.6.0 GTK3 AppImage.  When I tried to run it, I get:

```
/tmp/.mount_PrusaSKTtEzs/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwayland-client.so.0: cannot open shared object file: No such file or directory
```

Tried downloading the `.tar.bz` binary and running `prusa-slicer`, to no avail:

```
(Containing directory censored)/PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220/bin/prusa-slicer: error while loading shared libraries: libwayland-client.so.0: cannot open shared object file: No such file or directory
```

Tried the GTK2 AppImage as well:

```
/tmp/.mount_PrusaSKh6Ytx/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwayland-client.so.0: cannot open shared object file: No such file or directory
```

### Project file & How to reproduce

No project file.  Program refuses to start.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220 / PrusaSlicer-2.6.0+linux-x64-GTK2-202306191220

### Operating system

Gentoo Linux

### Printer model

MakerFarm i3v and custom CoreXY machine

kpodvolecky, 2023-06-26:
I have same issue. Workaround was to install dev-libs/wayland. It has small footprint, provides wayland-client libraries and does'n interfere with my existing X11 setup (so far).
But I was surprised by this error too.

KyleMaas, 2023-06-26:
@kpodvolecky
Thanks - that does seem to work as a workaround.  The core problem remains, however, that that library should probably be included in the AppImage.

closed 2026-09-29 by prusaslicer-bot[bot]