prusa3d/PrusaSlicer #10890: Hard Crashes when application is idle

arshish1612, 2023-06-23:
### Description of the bug

Leaving the application open for extended periods causes a crash. Crash Log below.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [73556]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0+MacOS-x64-202306191415 (PrusaSlicer-2.6.0+MacOS-x64-202306191415)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               503

Date/Time:             2023-06-23 22:32:14.1292 +0530
OS Version:            macOS 13.4.1 (22F82)
Report Version:        12
Anonymous UUID:        B2BEC73D-88BC-4331-AA92-941A726B5E63

Sleep/Wake UUID:       4A0E0FC9-5F5C-4809-9BB7-8019ECCF0B4C

Time Awake Since Boot: 43000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [73556]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 105553518919664
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x10129ee54 Slic3r::GUI::ObjectLayers::sys_color_changed() + 148
1   PrusaSlicer                   	       0x1011cd784 Slic3r::GUI::Sidebar::sys_color_changed() + 240
2   PrusaSlicer                   	       0x1011fef84 Slic3r::GUI::Plater::sys_color_changed() + 40
3   PrusaSlicer                   	       0x1011b2c78 Slic3r::GUI::MainFrame::on_sys_color_changed() + 60
4   PrusaSlicer                   	       0x102e2a1f0 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
5   PrusaSlicer                   	       0x102e29ebc wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
6   PrusaSlicer                   	       0x102e29ce4 wxEvtHandler::ProcessEvent(wxEvent&) + 56
7   PrusaSlicer                   	       0x102e2a39c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
8   PrusaSlicer                   	       0x102d62320 -[wxNonOwnedWindowController windowDidChangeBackingProperties:] + 400
9   CoreFoundation                	       0x1810a3180 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 148
10  CoreFoundation                	       0x18113eeec ___CFXRegistrationPost_block_invoke + 88
11  CoreFoundation                	       0x18113ee34 _CFXRegistrationPost + 440
12  CoreFoundation                	       0x1810744cc _CFXNotificationPost + 704
13  Foundation                    	       0x181fd28f4 -[NSNotificationCenter postNotificationName:object:userInfo:] + 88
14  AppKit                        	       0x184aa5ac8 -[NSWindow _postWindowDidChangeBackingPropertiesAndDisplayWindowForPreviousBackingScaleFactor:previousColorSpace:] + 164
15  AppKit                        	       0x1844952e4 __67-[NSWindow _updateSettingsSendingScreenChangeNotificationToScreen:]_block_invoke + 448
16  AppKit                        	       0x1843115f8 NSPerformVisuallyAtomicChange + 108
17  AppKit                        	       0x184495118 -[NSWindow _updateSettingsSendingScreenChangeNotificationToScreen:] + 96
18  AppKit                        	       0x184312ad8 -[NSWindow _setFrameCommon:display:fromServer:] + 1856
19  AppKit                        	       0x184a9f7e8 -[NSWindow(NSScreenLayout) _setFrame:fromAdjustmentToScreen:anchorIfNeeded:animate:] + 576
20  AppKit                        	       0x1845b6400 -[NSWindow _adjustWindowToScreen] + 1100
21  AppKit                        	       0x1845d5ce0 -[NSWindow _displayChangedSoAdjustWindows:] + 100
22  AppKit                        	       0x184d42bd8 ___NSApplicationPerformScreenInvalidationReactions_block_invoke + 72
23  AppKit                        	       0x1842c9360 -[NSApplication enumerateWindowsWithOptions:usingBlock:] + 236
24  AppKit                        	       0x184d426ec ___NSApplicationReactToScreenInvalidation_block_invoke + 848
25  AppKit                        	       0x184d420ac _NSApplicationReactToScreenInvalidation + 292
26  AppKit                        	       0x184d42284 -[NSApplication(ScreenHandling) _reactToDisplayChanges:] + 372
[... 461 lines cut ...]
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.

### Project file & How to reproduce

Just leave the application running and idle for a while with an object sliced

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

Macbook Air M1 - 13.3.1 (a) 

### Printer model

FLSUN V400

lukasmatena, 2023-06-28:
Thanks, reproduced. This is not a new issue in 2.6.0. In fact, it has been there since 2.4.0, when we implemented dark mode. It is also not macOS specific, the same happens on Windows.

The backtrace you provided helped greatly to find the problem, which only happens in quite specific scenario. It crashes when you have Height Range Modifier opened, a specific layer selected and you change the color theme of your system. There is a bug in the code that updates the controls in the object list. I assume that you have automatic color scheme enabled in your system, which gives the impression that it crashes after some time, but it is not the case. Can you confirm that?

KaiCollins1, 2023-07-07:
Just had this problem, good to know that this is a known bug. Luckily, I didn't lose any progress on anything important.

I feel as if some sort of warning about this known bug should be included. Dark mode is marked as experimental, but if there is a known unsolved (?) bug that causes crashes, an indication of that would be convenient.

lukasmatena, 2023-07-08:
@KaiCollins1 We will fix the bug in the next release, there is little else that we can do.

arshish1612, 2023-07-12:
> Thanks, reproduced. This is not a new issue in 2.6.0. In fact, it has been there since 2.4.0, when we implemented dark mode. It is also not macOS specific, the same happens on Windows.
>
> The [...]

No - I have Light mode enabled and not Automatic mode.

ctilley83, 2023-07-13:
Same crash. Mine doesn't seem related to dark mode either. It's happened at me during daytime hours and not during a switch from light to dark. I do not recall having this crash until 2.6.0 final, although I may have just gotten lucky with the betas. I can post the trace the next time it happens.

lukasmatena, 2023-07-24:
@arshish1612 @ctilley83 
Thanks for the reply. That is a bit unexpected, because the backtrace above leads directly into the "mode has changed" handler. However, that event may be emitted in some other scenario. Are you guys able to test the following build? It is 2.6.0 plus a single commit fixing the crash that I was able to reproduce ([52bc770](https://github.com/prusa3d/PrusaSlicer/commit/52bc770ebf3f5e28044c82108ef071876afb32d3)).

The build to test: [PrusaSlicer-2.6.0+1-lm-10890-MacOS-universal-g52bc770eb](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0+1-lm-10890-MacOS-universal-g52bc770eb-202307240753.dmg)

Thanks.

arshish1612, 2023-07-25:
Testing.. Will update

arshish1612, 2023-07-26:
Been running for 7 hours now.. No Crash as yet

lukasmatena, 2023-07-26:
@arshish1612 Thanks. I don't know how often were you experiencing the crashes before, but I take it as a confirmation that the fix solved the problem. If you notice the crash in that build later, please, let us know.

arshish1612, 2023-07-26:
@lukasmatena Looking good. No crash as yet. App still open since last night.

KaiCollins1, 2023-08-18:
Is there a fix that I can implement in Windows until a patch is next released? I've just been closing PrusaSlicer every 10 minutes or so when using it, which becomes cumbersome after a while.

lukasmatena, 2023-11-15:
This is fixed since [2.6.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1). Closing.

closed 2023-11-15 by lukasmatena