prusa3d/PrusaSlicer #10891: G Code generation stuck at 70%

yasaslive, 2023-06-23:
### Description of the bug

Slicer does not create support material and stuck at 70%. CPU usage is between 25-45% and when try close slicer crashes.

### Project file & How to reproduce

1. Open file. 
2. Place object and slice.
[image]
 
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/11851391/test.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+win64-202306191221

### Operating system

Windows 11 22H2

### Printer model

Not relevant

yasaslive, 2023-06-23:
This issue cannot be found in alpha versions. Only in beta and release version. 
Tested version Release: PrusaSlicer-2.6.0+win64-202306191221
Tested version Alpha: PrusaSlicer-2.6.0-alpha6+win64-202303311519

foreachthing, 2023-06-23:
Maybe this helps: https://github.com/prusa3d/PrusaSlicer/issues/10858 ?

bubnikv, 2023-08-29:
The issue was indeed connected to raft. I twill be fixed in PrusaSlicer 2.6.2 final.

closed 2023-08-29 by bubnikv