prusa3d/PrusaSlicer #10893: Crash during slicing

sbprusa, 2023-06-24:
### Description of the bug

when slicing the part prusaSlicer2.6 just always crashes

### Project file & How to reproduce

[LP02BodyV1.0.10.zip](https://github.com/prusa3d/PrusaSlicer/files/11856776/LP02BodyV1.0.10.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 win64

### Operating system

win11

### Printer model

mk4

sbprusa, 2023-06-25:
...I renamed the "NahimicOSD.dll", afterwards prusa works stable

tracrouch, 2023-06-28:
having same issue

lukasmatena, 2023-06-29:
@tracrouch Your issue may be something different. Does PrusaSlicer complain about the Nahimic library on startup? If it doesn't, I suggest you file a separate bug report and provide more details.

@sbprusa If PrusaSlicer gave the hint about the Nahimic library and renaming it helped, I guess we can as well close this issue, because there is probably nothing else we can do about it.

Jan-Soustruznik, 2023-07-10:
Hello, @tracrouch , 

Thank you for your feedback. Problem is duplicated by https://github.com/prusa3d/PrusaSlicer/issues/5573 

Issue closed

closed 2023-07-10 by Jan-Soustruznik