prusa3d/PrusaSlicer #10898: Save Setting in Multiple Presets

AustinTylerDean, 2023-06-24:
**Is your feature request related to a problem? Please describe.**

PrusaSlicer is my second foray into slicers, coming from Cura. The profile management divided between print/filament/printer is fundamentally a game changer. Alas, not all is perfect! As I tune my Voron 2.4, I get deeper into my understanding of a setting, _and find that MANY times I am changing a setting that would be ideal in ALL of my profiles, which could be any print/filament/printer setting._

**Describe the solution you'd like**

I would like a **hotkey that I could hold while either clicking out or pressing enter** after entering a parameter to have a pop up check box list window show all my (print or filament or printer) profiles in a list with them already checked, and a submit or cancel button at the bottom. This would change the current setting in all the profiles.

**Describe how it would work**

Hold hotkey-N while either clicking out of the newly entered parameter or pressing enter to have said window pop up.

**Describe alternatives you've considered**

I have not seen this feature on any slicer, though my experience is limited.

**Additional context**
N/A

Tupson444, 2023-06-25:
There is a workaround for this. Although not exactly quick, it is faster than manually changing each preset (when there are many):

Open .ini files and compare old and new settings (on Windows it's probably located in C:\Users\user name\AppData\Roaming\PrusaSlicer\print). Then use "Find in files" option in Notepad++, for example:

find `seam_position = nearest`
replace with `seam_position = aligned`
and select appropriate directory

Note: it's a good idea to make backups of all settings from time to time, especially if there is a risk of braking something.

---

However, it would be nice to be able to do it right in the slicer, quickly.

AustinTylerDean, 2023-06-25:
> There is a workaround for this. Although not exactly quick, it is faster than manually changing each preset (when there are many):
>
> Open .ini files and compare old and new settings (on Windows it's [...]

Thanks for the "workaround", however considering my skill level, it would probably be faster for me to just switch to all the profiles and copy and paste my parameters each time I change something... Also, I would hate if someone with the technical chops read this reply and thought of this as less of a priority because a "workaround" was mentioned!

If this could be built, I imagine the feature would be quite unique and very much desired by those running rather dynamic setups. Here's to hoping! Cheers!

Tupson444, 2023-06-27:
>Also, I would hate if someone with the technical chops read this reply and thought of this as less of a priority because a "workaround" was mentioned!

I haven't thought of that. I really hope not, because the workaround is complicated, while the aim here is to make things simpler and more versatile.

Tupson444, 2026-09-12:
Still relevant, especially since the workaround I used will probably be useless in 3.0.0 because the profiles are saved in different kind of file.