prusa3d/PrusaSlicer #10899: Nozzle hitting parts M600 PS 2.6

x7moodx94, 2023-06-25:
### Description of the bug

Hi everyone, first of all thanks for the greatest slicer to be used; with no further introductions
I've been using PS recently 2.5 was my first version I use it only for multicolor prints with the use of M600 in gcode 

The problem happened  to me twice yesterday and when the it was beta 3 or 4

Back then I thought it was some setting problem but when it happened yesterday I thought I should post about it

To describe the problem, I print multiple pieces at ones every time, the print went really smoothly until the last layer of the last part before filament change
when the layer was done the print head went into the the printed part, went to X0, Y0 while hitting every thing in the way, I changed the filament color and pressed resume thinking that one part out 3 I can deal with it, but after I finished the swap the print head went into the parts before going to the wipe tower

I will share 2 gCodes one sliced with 2.5.2 the other 2.6 I took the settings out of the old one changed in the temperature and flow only

My printer is CR-10 Th3d EZBoard and Marlin 2 from TH3d everything else is mostly stock
[image]
[image]
[Gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/11858869/Gcodes.zip)

### Project file & How to reproduce

[Gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/11858871/Gcodes.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.60 Final

### Operating system

Windows 10

### Printer model

Creality CR-10

BogdanPNT, 2023-06-25:
I can confirm this issue happened to me as well. Using an Ender 3, was printing multi color succesfully using the M600 double extruder trick. Upgraded to 2.6 and it happened exactly like he said: Prints perfectly, nozzle goes into the part, lifts up, you do the filament change and than in plunges back into the part.

x7moodx94, 2023-06-25:
> 

So this is happening and I thought I was alone

rtyr, 2024-02-18:
The M600 placement was improved in 2.7.2-alpha2. It should fix your problem, even though it was mostly caused by standard firmware behavior (returning to the place, where M600 command was initiated).

closed 2024-02-18 by rtyr