prusa3d/PrusaSlicer #10901: URI Too Long

Uppermill, 2023-06-25:
### Description of the bug

SAfter implementing PrusaSlicer 2.6.0 on Win64, after slicing I try and send to MK4 printer using Web Link and get the following message
[image]

### Project file & How to reproduce

See error

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 64

### Printer model

MK4

Elfe, 2023-07-09:
Same issue here. Does the URI contain the gcode filename?
I used an other (shorter) filename in the upload dialog and it worked out fine.

Slicer: PrusaSlicer-2.6.0+linux-x64-GTK3-202306191220.AppImage
Printer: Prusa i3 MK4 with FW 4.7.0 over WiFi (PrusaLink)

Uko, 2023-07-11:
+1 This is very annoying. Definitely not critical, as I can rename the file in the pre-upload dialog, but I suppose there are many ways to solve this

iccor56, 2023-09-07:
i have the same issue. if it worked with the last firmware it should work with this firmware especially since that is the default name the prusa software uses. 

so i question what has changed......

[15.5.15.](https://httpwg.org/specs/rfc9110.html#rfc.section.15.5.15) [414 URI Too Long](https://httpwg.org/specs/rfc9110.html#status.414)

The 414 (URI Too Long) status code indicates that the server is refusing to service the request because the target URI is longer than the server is willing to interpret. This rare condition is only likely to occur when a client has improperly converted a POST request to a GET request with long query information, when the client has descended into an infinite loop of redirection (e.g., a redirected URI prefix that points to a suffix of itself) or when the server is under attack by a client attempting to exploit potential security holes.

sadovsf, 2023-09-08:
You are digging into it too deep and trying to apply pc world explanation of this error code to embedded system (prusa printer). It's much more simple. There are limited resources on printer and to avoid stalling printing (probably) or other issues. Printer has limit to url length. Unfortunately url with default format of file name is too long and printer refuses to interpret it hence the returned error code. Limit on its own could be fairly simple fix but related implications of such change may be bigger than we can expect. It's reported, be patient. Sw development is not easy 😉

iccor56, 2023-09-09:
no i am applying HTTP error codes to the issue. http is http no matter what platform. and while it has been reported (posted here) it has not yet garnered the attention of the programmers so your comment is both inaccurate and the antitheses of the point of this forum.

sadovsf, 2023-09-10:
Http is standard and as everything depends on actual implementation. I work as c++ programmer over 10y both pc and ebbeded. Yes there are huge differences.

JacobChrist, 2023-09-21:
This isn't an issue with the Prusa Slicer but it would be nice if the slicer warned you that it would be an issue and gave you the opportunity to rename.  Since I don't know the max length that printer firmware will accept I often end up shortening the file name a couple of times before the printer will take it.

Jan-Soustruznik, 2023-09-21:
Displayed ERR message is from FW. Issue was moved to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3252 

I closed this ticket.

JacobChrist, 2023-09-21:
@Jan-Soustruznik The firmware already displays an error message.  What would be nice is if the Slicer warns you that the file name is too long for the printer before you spend time uploading the file and it fails.