prusa3d/PrusaSlicer #10902: Filament Settings with Multiple Prusa Printers

Uppermill, 2023-06-25:
I have  a MK3S+, MK4 and Mini+ printers. When I change printer on PrusaSlicer, I always get a default set of filament settings, rather than the last ones used. When I select my MK3+ with the MMU, I have to set the filament type for each of the five filaments available.

If PrusaSlicer remembered the last filament settings, or had defaults that could individually be set for each printer, then this would be a big help.

neophyl, 2023-06-25:
The common way to do this is to save a 3mf project file for each common used setup.  As a 3mf saves all the settings and profiles in use  including your filament selections for your MMU it allows you to load in exactly what you want and you can have as many as you want.

Uppermill, 2023-06-25:
Thanks - I will try and use 3mf files.

Jan-Soustruznik, 2026-01-20:
closed as duplicate