prusa3d/PrusaSlicer #10903: Exporting of sliced model is not working

chris0550, 2023-06-25:
### Description of the bug

I sliced a model (https://cults3d.com/de/modell-3d/haus/bean-tray-extend-for-magnifica-s ) but the export to gcode is not working. The export never finishes. Other models are working.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

ubuntu with wine

### Printer model

Kobra 2

foreachthing, 2023-06-26:
Please upload your .3mf _(zip it first - .3mf can't be uploaded)_ file. As settings are the culprit, most of the time.

chris0550, 2023-06-26:
Working with 2.6.0

closed 2023-06-26 by chris0550

kubispe1, 2024-03-05:
Hi @foreachthing, please can you contact me to my mail? 
petr.kubis@prusa3d.cz
Thanks and have a nice day.