prusa3d/PrusaSlicer #10909: Linux GUI cutoff does not allow for Ramming Customizations

aerialviews007, 2023-06-27:
### Description of the bug

I was attempting to use a guide to tweak my filament profile for the MMU2S .  The guide is for PrusaSlicer for Windows but I have Fedora 38 (although I have also tried on other linux distros).  The ability to add time to the ramming customization is cut off and there is no ability to change the numbers manually.   Please see screenshot:

[image]

### Project file & How to reproduce

1. Open PrusaSlicer
2. Click Filament Settings
3. Enter Expert Mode
4. Click Advanced
5. Click Ramming Settings under Ramming parameters

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 but also exists for earlier versions

### Operating system

Fedora 38 Flatpak and AppImage

### Printer model

Prusa mk3s W/ MMU2S

khumarahn, 2023-07-28:
Same here, gentoo, appimage. Is there a workaround?

cheekylions, 2023-08-01:
I ended up saving a profile, opening that in PrusaSlicer on a Windows VM, changing the ramming time, then saving the profile and putting it back in my Linux profiles folder.

But to save some trouble, you can edit the .ini of your MMU2S profile and use one of the following settings:

For a 3.50 Ramming Time:
`filament_ramming_parameters = "130 120 2.70968 2.93548 3.32258 3.83871 4.54839 5.51613 6.58065 7.41935 7.77419 8 8.96774 10.5161 11.4516 11.5161| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6"`

For a 3.00 Ramming Time:
`filament_ramming_parameters = "130 120 2.70968 2.93548 3.32258 3.83871 4.54839 5.51613 6.58065 7.41935 7.70968 7.93548 9.09677 10.6452| 0.05 2.66451 0.45 3.05805 0.95 4.05807 1.45 5.97742 1.95 7.69999 2.45 8.1936 2.95 11.342 3.45 11.4065 3.95 7.6 4.45 7.6 4.95 7.6"`

pgrondek, 2023-08-19:
My workaround for this problem is to use GTK2 version
[image]

lukasmatena, 2023-10-20:
Closing as a duplicate of #6399. It will be fixed in the upcoming release.

closed 2023-10-20 by lukasmatena