prusa3d/PrusaSlicer #10910: Label Objects option should label wipe tower

scottgrim, 2023-06-27:
**Is your feature request related to a problem? Please describe.**
With some features in klipper, object labels are used to determine the actual utilized print area so that only part of the bed requires ABL and the purge can occur next to the object (https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging), for example.  Wipe tower is not labeled as an object so isn't included in these algorithms.

**Describe the solution you'd like**
Label wipe tower just as any other object in the print area.

**Describe alternatives you've considered**
Alternately, add substitutions that expand to the X/Y coordinates of the 4 corners of the wipe tower so that they can be passed to a macro in start g-code.

kyleisah, 2023-07-14:
@scottgrim thanks for opening this issue here, I would really like this to happen so I can make KAMP accomodate MMU setups. I'm digging around in prusaslicer and can really only find a couple things that could possibly define where the wipe tower is. If I can force those values into positions that can be passed, you should be able to define an exclude object from it. Looks like no bites on this issue yet. Hopefully soon.

kyleisah, 2023-07-14:
I'll be tracking my progress on a stop-gap at KAMP issue #121

baremetaldude, 2025-04-22:
For now I can move wipe tower as close as possible to printed object and pray that everything will be fine, but it would be nice to have additional exclude definition
EXCLUDE_OBJECT_DEFINE NAME='Wipe-Tower' CENTER=... POLYGON=[[...]]
for ABL to work

kyleisah, 2026-09-14:
Bump. All objects that are present on any build plate should be labeled, which includes a wipe tower. The klipper firmware can build the bounds for a bed mesh based off of labeled objects present in the gcode. Right now, however, the wipe tower is not labeled as an object, so it is missed, and may cause first layer issues for users who slice with PrusaSlicer and utilize object-based bed meshes.