prusa3d/PrusaSlicer #10911: Arrange function not working with off-bed objects

wouldstain, 2023-06-27:
### Description of the bug

Arrange function fails after making off-bed object printable. Error message is "Arrangement ignored ..." (see attached screenshot) This may be related to the number of objects in the file but it happens after toggling an object on and then hitting either "A" or clicking the "arrange" button. Working without issue in previous version.
[boxes.zip](https://github.com/prusa3d/PrusaSlicer/files/11876828/boxes.zip)

### Project file & How to reproduce

toggle object off
toggle different object on
select "arrange"

Error does not occur 100% of the time.

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 Windows

### Operating system

Windows 11

### Printer model

Railcore 300 ZLT

Jan-Soustruznik, 2023-07-31:
Hello, @wouldstain , 

Thank you for your feedback. We can reproduce issue on our side. We will try to fix in some next releases. 

Thank you, have a nice day

issue: SPE-1830

FidelCapo, 2023-08-21:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-08-21 by FidelCapo