prusa3d/PrusaSlicer #10912: Prusa slicer showing errors when adding bunny ears

thefett1, 2023-06-27:
### Description of the bug

Latest version is throwing up error ( conflicts in gcode paths ) when using bunny ears on any object. Is there a way to permanently disable that warning? I use bunny ears all the time.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-Max64

### Operating system

10.15.7 mac

### Printer model

prusa mk3s

rtyr, 2023-06-27:
Please provide your 3MF project.

shledge, 2023-06-27:
They need to be merged with the object itself.

ulab, 2023-07-18:
I wonder if an "auto-merge overlapping parts"-feature would be an idea?

[Optional in case it makes sense for specific cases?]

Reference: https://twitter.com/PrusaSlicer/status/1681236516985614336

rtyr, 2023-11-23:
No response. Closing.

closed 2023-11-23 by rtyr