prusa3d/PrusaSlicer #10913: [Feature Request] As a user, I would like to be able to have only one perimeter on visible bottom/top layers

meltiseugen, 2023-06-27:
**Is your feature request related to a problem? Please describe.**
Having only one perimeter line on the first bottom and last top layer improves the aspect of the printed part. Especially when there are flat faces involved (This also is applied for intermediate bottom/top sections).

**Describe the solution you'd like**
I would like 2 checkboxes: 
* one for activating the feature on the first bottom layers
* one for activating the feature on the last top layers

**Describe how it would work**
If the 2 checkboxes are active, then the first/last bottom/top layer will have only one perimeter, ignoring the defined perimeters option.
All the other layers are printed using the normal perimeters value.

**Describe alternatives you've considered**
An alternative is printing the entire part with only one perimeter everywhere, but it will not be a very strong part.

**Additional context**

An example on how it should look:
[image]

An example on how the rest of the layers should look
[image]

An example of the top section, prior to the last layer of the top section
[image]

foreachthing, 2023-06-27:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7986 and many more.

neophyl, 2023-06-27:
You can also use a height range modifier to do this, easily with the bottom layer.  Its harder with the top layer depending on how many top surfaces there actually are but for simpler geometry it is possible.

rtyr, 2023-06-27:
Closing as a duplicate. We are planning to add this feature.

closed 2023-06-27 by rtyr

erlis, 2023-10-22:
please add this soon!