prusa3d/PrusaSlicer #10915: Set external/network directory for configuration

shledge, 2023-06-27:
I run PrusaSlicer on multiple computers at home - I have a NAS which stores things like my 3D files and configuration, and would love if Prusaslicer had the ability to link to the configuration on the NAS itself, meaning all computers with Prusaslicer will be in sync in terms of settings etc.

mix579, 2023-06-27:
You can do that today. Copy the configuration directory from one of your computers to your NAS. Then on each computer replace the local config directory with a symlink to the directory on the NAS. I've been running this on multiple macs and a windows machine for years, albeit using Onedrive

shledge, 2023-06-27:
Doesn't work for me sadly

sarusani, 2023-06-28:
> Doesn't work for me sadly

Symlinks normally work fine on Windows/MacOS/Linux. What's not working for you?

Another way is to use the --datadir parameter when opening the slicer.
For example:
prusa-slicer.exe --datadir Z:\slicer-settings

shledge, 2023-06-28:
If I make the config directory a symlink, prusaslicer refuses to load

sarusani, 2023-06-28:
I'm on windows 10. This is exactly how I made it work on my setup: (Example works for rc & final releases. Alpha & beta use a different folder.)
First I moved the Folder "C:\Users\[MyUserName]\AppData\Roaming\PrusaSlicer" to my NAS.
Then I created the Symlink like this:
`mklink /D C:\Users\[MyUserName]\AppData\Roaming\PrusaSlicer \\[serverIP]\[share]\PrusaSlicer\`

If this does not work for you (assuming your on windows), could you run: "fsutil behavior query symlinkevaluation" and post the outpup?
Should be something like:
>Local to local symbolic links are enabled.
Local to remote symbolic links are enabled.
Remote to local symbolic links are disabled.
Remote to remote symbolic links are disabled.

shledge, 2023-06-28:
Again, I'm on MacOS.

sarusani, 2023-06-28:
> Again, I'm on MacOS.

Sorry, I totally missed that... My bad.

Have you tried using the --datadir method?

mix579, 2023-06-28:
Here's my recipe for macOS:

mix579, 2023-06-28:
Here's my recipe for macOS:

* Go to Prusaslicer's Help menu and select Show Configuration Folder
* Go up one level and note the location of it. It's usually /Users/username/Library/Application Support/PrusaSlicer (alphas and betas have different names).
* Create a backup copy, just in case.
* Copy the configuration folder to your network drive and delete the original. Note the new pathname, in my case /Users/username/OneDrive/3D Printing/Software/PrusaSlicer/PrusaSlicer
* Open a terminal window.
* Create a symlink:
```
ln -s /Users/username/OneDrive/3D Printing/Software/PrusaSlicer/PrusaSlicer /Users/username/Library/Application Support/PrusaSlicer

``
```