prusa3d/PrusaSlicer #10916: Crash when auto arrange

Teegan8or, 2023-06-28:
### Description of the bug

I have a custom bed geometry (which I assume is the problem) and when I go to auto-arrange my parts on the bed, Prusa Slicer crashes.

https://github.com/prusa3d/PrusaSlicer/assets/137978853/00887969-f612-4cbf-bef5-1d7fdd4c3f1a

### Project file & How to reproduce

1. Have a custom bed geometry
2. Import models
3. arrange bed

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 10

### Printer model

Tevo Tornado

foreachthing, 2023-06-28:
Did you load your stl as MODEL or SHAPE?
I had issues with MODEL as well and loaded it as SHAPE and it worked.
[image]

foreachthing, 2023-06-28:
Until now.... !!! Ok! Same issue with shape!

lukasmatena, 2023-06-28:
Thanks, reproduced. We will fix it.
SPE-1800

closed 2026-09-29 by prusaslicer-bot[bot]