prusa3d/PrusaSlicer #10917: bottom of Support is outside printable area

DD8HL, 2023-06-28:
**Is your feature request related to a problem? Please describe.**
when using supports (especially the great new organic support) and the model ist pretty tight to the maximum printable size, the base of the support is outside the printbed and an arror occours when slicing.

**Describe the solution you'd like**
especially for organic supports it would make sense to let them start inside the printable areas to avoid the error

**Describe how it would work**
a support blocker for the outside of the printbed

**Describe alternatives you've considered**
and option to just plain cut away everything thats outside the printable area (minus a little tolerance say 1mm)

**Additional context**

[image]

neophyl, 2023-06-28:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9569 which is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/849 and probably many more.

Jan-Soustruznik, 2024-04-12:
Supports that are generated out of bed in the known issue. https://github.com/prusa3d/PrusaSlicer/issues/9569 and original https://github.com/prusa3d/PrusaSlicer/issues/849

closed 2024-04-12 by Jan-Soustruznik