prusa3d/PrusaSlicer #10920: Add config option to zoom in on cursor by default

beelsebob, 2023-06-28:
**Is your feature request related to a problem? Please describe.**
Not really - just an annoyance

**Describe the solution you'd like**
PS 2.6 added the ability to zoom in on the cursor instead of the centre of the screen by holding shift down while zooming.  Can we please have the option to make that the default zooming mode (without holding shift).

chrisgp123, 2023-11-04:
Agree! Also, holding shift does not seem to respect the Reverse Scroll direction. Normally I scroll forward to zoom in but if I hold shift, I have to scroll backwards to zoom out.

mdeeter, 2023-12-12:
I'm not seeing this working correctly... in previous version, you could hold `shift` and zoom to cursor... but I'm not seeing this work at the moment.

```
PrusaSlicer
Version:   2.7.0+MacOS-arm64
Build:     PrusaSlicer-2.7.0+MacOS-arm64-202311231454

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 14.0 (Build 23A344)
Total RAM size [MB]: 25,770MB
OpenGL installation
GL version:   4.1.0
Profile:      Core
Vendor:       Apple
Renderer:     Apple M2
GLSL version: 4.10.0
Textures compression:       Enabled
```

neophyl, 2023-12-12:
its working on the windows release, at least for win 10

smurfix, 2024-04-19:
Works in the Linux (Debian) release.

I had no idea that pressing Shift to zoom to the cursor is even an option before I found this issue, and frankly I'd like a config setting that reverses this. Or you could set the default zoom-to-cursor unconditionally — I have no idea why anybody would _ever_ want to zoom to center-of-screen instead of to cursor position …

chrisgp123, 2026-09-18:
This is still not fixed in 3.0.0-alpha11. In fact, Shift+scroll doesn't work either.

Jan-Soustruznik, 2026-10-22:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #15619.

closed 2026-10-22 by Jan-Soustruznik