prusa3d/PrusaSlicer #10922: Can't run in Raspberry pi 4B

daizhuo, 2023-06-29:
### Description of the bug

Can't open the Appimage.
I can just open 2.4.2, any version newer than it will NOT work.

### Project file & How to reproduce

**

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+linux-armv7l-GTK2-202306191220.AppImage

### Operating system

raspberry pi 4B

### Printer model

i3 Mk3s+

Jan-Soustruznik, 2023-07-12:
Hello, @daizhuo , 

Thank you for report. Yes, released build isn't compatible with 64-bit. We will try to fix in next release. 

issue: SPE-1812

Have a nice day

Jan-Soustruznik, 2023-07-18:
Hello, @daizhuo , 

We compiled compatible build to 64-bit. Build can be download from final release 2.6.0, link: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0

I closed issue as solved. 

thank you, have a nice day

closed 2023-07-18 by Jan-Soustruznik