prusa3d/PrusaSlicer #10923: Crash on exporting simplified .stl

larsinka, 2023-06-29:
### Description of the bug

Prusaslicer crashes when I try to export an .stl file that I simplified before.

I expected no crash.

Reproduces every time.

[Extrudinaire.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11905455/Extrudinaire.3mf.zip)

### Project file & How to reproduce

1. Right-click on the model.
2. Click on 'Export Object as STL/OBJ'
3. Choose name
4. Click 'save'
5. Crash

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

MacOS Ventura 13.4

### Printer model

Mk3S+

larsinka, 2023-06-29:
MacOS Crash-log:

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [6022]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0+MacOS-x64-202306191415 (PrusaSlicer-2.6.0+MacOS-x64-202306191415)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-06-29 14:06:10.1470 +0200
OS Version:            macOS 13.4 (22F66)
Report Version:        12
Anonymous UUID:        B70013AA-93BF-205F-CFB3-B1938A245007

Sleep/Wake UUID:       D2586276-DD9E-49DF-A80B-C75A8005EFDB

Time Awake Since Boot: 61000 seconds
Time Since Wake:       1755 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000001138617ff0
Exception Codes:       0x0000000000000002, 0x0000001138617ff0

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [6022]

VM Region Info: 0x1138617ff0 is in 0x1000000000-0x7000000000;  bytes after start: 5240881136  bytes before end: 407075979279
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)         fc0000000-1000000000   [  1.0G] ---/--- SM=NUL  ...(unallocated)
--->  GPU Carveout (reserved)    1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
      GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x1011f5b58 unsigned long CGAL::Polygon_mesh_processing::volume_connected_components<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick>>, CGAL::Constant_property_map<CGAL::SM_Face_index, unsigned long>, CGAL::Named_function_parameters<bool, CGAL::internal_np::i_used_as_a_predicate_t, CGAL::Named_function_parameters<bool, CGAL::internal_np::do_orientation_tests_t, CGAL::Named_function_parameters<bool, CGAL::internal_np::all_default_t, CGAL::internal_np::No_property>>>>(CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick>> const&, CGAL::Constant_property_map<CGAL::SM_Face_index, unsigned long>, CGAL::Named_function_parameters<bool, CGAL::internal_np::i_used_as_a_predicate_t, CGAL::Named_function_parameters<bool, CGAL::internal_np::do_orientation_tests_t, CGAL::Named_function_parameters<bool, CGAL::internal_np::all_default_t, CGAL::internal_np::No_property>>> const&) + 1292
1   PrusaSlicer                   	       0x1010e20b4 Slic3r::MeshBoolean::cgal::does_bound_a_volume(Slic3r::MeshBoolean::cgal::CGALMesh const&) + 60
2   PrusaSlicer                   	       0x100c9da70 std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>, std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&)::'lambda'(std::__1::__wrap_iter<Slic3r::csg::CSGPart*>&)>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&, std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&)::'lambda'(std::__1::__wrap_iter<Slic3r::csg::CSGPart*>&)&&)::'lambda'(unsigned long)::operator()(unsigned long) const + 112
3   PrusaSlicer                   	       0x100c9d954 void tbb::detail::d1::dynamic_grainsize_mode<tbb::detail::d1::adaptive_mode<tbb::detail::d1::auto_partition_type>>::work_balance<tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, void Slic3r::execution::Traits<Slic3r::ExecutionTBB, void>::for_each<unsigned long, std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>, std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&)::'lambda'(std::__1::__wrap_iter<Slic3r::csg::CSGPart*>&)>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&, std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&)::'lambda'(std::__1::__wrap_iter<Slic3r::csg::CSGPart*>&)&&)::'lambda'(unsigned long)&>(Slic3r::ExecutionTBB const&, std::__1::__wrap_iter<Slic3r::csg::CSGPart*>, std::__1::__wrap_iter<Slic3r::csg::CSGPart*>, std::__1::__wrap_iter<Slic3r::csg::CSGPart*> Slic3r::csg::check_csgmesh_booleans<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>>(Slic3r::Range<std::__1::__wrap_iter<Slic3r::csg::CSGPart*>> const&)::'lambda'(std::__1::__wrap_iter<Slic3r::csg::CSGPart*>&)&&, unsigned long)::'lambda'(std::__1::__wrap_iter<Slic3r::csg::CSGPart*> const&), tbb::detail::d1::auto_partitioner const>, tbb::detail::d1::blocked_range<unsigned long>>(std::__1::__wrap_iter<Slic3r::csg::CSGPart*>&, std::__1::__wrap_iter<Slic3r::c [... cut ...]        8        1 \n__LINKEDIT                       825.0M        5 \n__OBJC_RO                         66.3M        1 \n__OBJC_RW                         2011K        1 \n__TEXT                           600.7M      550 \ndyld private memory                320K        4 \nmapped file                      190.6M       39 \nshared memory                     1024K       24 \n===========                     =======  ======= \nTOTAL                              4.5G     3352 \nTOTAL, minus reserved

kubispe1, 2023-07-10:
Hi, we have investigated this issue already and we are trying to fix it:
SPE-1801

madhephaestus, 2023-09-29:
This seems related to https://github.com/prusa3d/PrusaSlicer/issues/11199

lukasmatena, 2023-10-04:
We already have a fix for this problem (unrelated to #11199). It will be ok in the next release.

lukasmatena, 2023-11-15:
This is fixed as of [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.