prusa3d/PrusaSlicer #10926: New 2.6 ensure vertical shell thickness always adds extra thickness

AudreyAP, 2023-06-29:
### Description of the bug

Ensure vertical shell thickness **always** adds extra perimeters, even if they are very clearly not needed. 

I've attached a 3mf file using the stock MK3s+ profile. I've set perimeter count to 20 to demonstrate the issue further. Setting any perimeter count whatsoever results in extra perimeters, even once the part is several inches thick. 

Screenshot of the example file, a cylinder rotated a bit, with 20 perimeters set:

[image]

This issue is present on all files. It is particularly egregious on 3dbenchy, where the new behavior drastically increases print time. 

### Project file & How to reproduce

[Ensure vertical thickness bug.zip](https://github.com/prusa3d/PrusaSlicer/files/11909730/Ensure.vertical.thickness.bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0+win64

### Operating system

Windows 10 

### Printer model

Prusa MK3s+

Tupson444, 2023-06-30:
This has already been reported multiple times, the "main" issue is #10102.

Since many users want to be able to turn this off, I hope the issue will be taken seriously for the next release.

AudreyAP, 2023-06-30:
This is a separate issue - being able to disable this feature is another thing.

What I'm describing is that the actual part thickness is being ignored. The extra perimeters are ALWAYS added, even when the part is inches thick. My understanding is that it is supposed to ensure a certain level of thickness, not just arbitrarily add extra perimeters at all times.

lucasavendano, 2023-06-30:
I think this is all related in some way, #10928 #10202

Skyhawky-NL, 2023-08-24:
I have the same thing happening with PrusaSlicer 2.6.0. I get unnecessary infill in 2.6.0 that gets worse the more bottom layers I add.

This is with PrusaSlicer 2.6.0
[image]
The purple infill on the wall is also in mid air.

And this is with PrusaSlicer 2.5.2  (with "Ensure vertical shell thickness" disabled)
[image]

The model is DragonScalePot from Printables (https://www.printables.com/model/353853-dragon-scales-pot-and-planter-vase-mode)

saveman71, 2023-11-22:
Thanks to a [workaround from @neophyl on the printables.com forums](https://forum.prusa3d.com/forum/prusaslicer/removal-of-solid-infill-on-a-part-with-zero-infill/#post-681172), I was able to remove completely the solid infill by adding a height modifier on the whole height except bottom and top, then setting these parameters

[image]

But I would vastly prefer an option to turn that off in the settings.

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena