prusa3d/PrusaSlicer #10935: Add option to disable DLL injection warnings

GuardianGuy, 2023-07-02:
**Is your feature request related to a problem? Please describe.**
It's quite annoying having this warning popping up every time I open prusaslicer or open stl's.  I get that it's necessary to have the warning, but it's frustrating that that only problem I have related to the nahimicOSD is the warning message.  Disabling NahimicOSD is not an option for me since the system equalizer in alienware command center needs it to function.
[image]

**Describe the solution you'd like**
I think a check box or extra button on the popup that says something like don't show again would work
something like this:
[image]

shisva, 2024-03-01:
Hi Prusa Team! Any news on this request ?

jh62, 2025-01-08:
ASAP

closed 2026-09-29 by GuardianGuy