prusa3d/PrusaSlicer #10936: Object part within negative volume detected as outside of print area

vovodroid, 2023-07-02:
### Description of the bug

Object with part within negative volume located outside of print bed it can't be sliced:

[image]

### Project file & How to reproduce

Make project as in screenshot.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0

### Operating system

Win10 x64

### Printer model

Biqu B1 SE

Jan-Soustruznik, 2023-07-10:
Hello, @vovodroid, 

Could you share a 3mf file? Because I dont see any issue. Thank you

vovodroid, 2023-07-10:
Sure, here is it. Nothing special, just move part of object inside negative volume out of bed:

[image]

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/12003045/Shape-Box.zip)

Jan-Soustruznik, 2023-07-12:
Yes, now I see what you means. If I have correct info, then boolean operations are performed during slicing. So this isn't issue. I add the "feature request" mark to your ticket. 

Thank you, for you feedback, have a nice day.

closed 2026-09-29 by prusaslicer-bot[bot]