prusa3d/PrusaSlicer #10937: vanishing model after cut

HermanDoberman, 2023-07-02:
### Description of the bug

when i am doing cut for a moment everything is all right but after couple of seconds my cut is vanishing (sorry for my english)
[image]
[image]

### Project file & How to reproduce

[Modular_Drawer-Box_with_hexagonal_pattern_2827563.zip](https://github.com/prusa3d/PrusaSlicer/files/11929580/Modular_Drawer-Box_with_hexagonal_pattern_2827563.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

windows 10

### Printer model

anycubic kobra neo

neophyl, 2023-07-02:
Unfortunately the files you have attached are just the model files.  Not your 3mf project file saved from PS.  When I load the model into PS and cut it cuts perfectly fine and will slice.
[image]
As you can see that works normally so its something else besides the models.

Its is quite often your setup that is the problem which is why a 3mf project files saved from PS is requested when you open an issue.  Or maybe its when you cut at a specific height ?

Jan-Soustruznik, 2023-07-12:
Hello, @HermanDoberman , 

Thank you for you report, but without your project file(3mf) we cannot investigate reported problem. Could you share your project file, please? Thank you, have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]