prusa3d/PrusaSlicer #10940: Detailed models - Searching support spots takes forever, even when supports are disabled.

viesturz, 2023-07-02:
### Description of the bug

Searching support spots step is very slow on detailed models and seems to be run even when supports are not requested.

### Project file & How to reproduce

Open this marlin: https://www.printables.com/model/5480-marlin-multi-material/files
Slice

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

win 11

### Printer model

voron 2.4

FidelCapo, 2023-07-25:
This issue should be fixed and fix will be part of the next release.
Closing.

Colboto, 2023-11-30:
I can verify that this bug is still present in version 2.7.0. Support generation will process even if no supports are requested and they are very slow on detailed models.

DanielDC88, 2023-12-01:
Sorry to add, it looks like Prusaslicer could be running out of memory at this stage. It’s using 10GB whilst calculating overhang perimeters.

Gryfang, 2023-12-29:
This is not completed. Prusa Slicer 2.7.1 hangs on "Searching Support Spots" if you turn on fuzzy skin at all, and takes, in my case, almost an hour to finish slicing. This is NOT a complex object. It was a box 20mm x 127mm x 177mm, and supports were turned off because they were not needed. I was using fuzzy skin outside with .05 and .05 settings. It puts a really nice texture on things.
I left it running at about half an hour, fired up Cura using the same settings, and even having to figure out how to NOT get the inside textured as this if part of a picture frame, I was still printing before Prusa even finished. I am still waiting for Prusa to finish and at the 3rd layer of my print. It's been well over 30 minutes since it got stuck. Why are you Searching Support Spots if I turned off supports?

Jan-Soustruznik, 2024-01-02:
Hello, @Gryfang ,

Could you share the project file (3MF), where the problem occure, please? Thank you

non-bin, 2024-01-13:
Support searching still happens when supports are off, even with "Alert when supports needed" off.

@Gryfang:
I can't reproduce the hanging problem, and fuzzy skin adds a little time to perimiter generation but doesn't seem to affect the support generation stage. I have a pretty beefy computer though, what are you using?

@DanielDC88:
Mine stays at a prety consistent 1GB RAM usage throughout, is that system usage, or just Prusa Slicer?
Could you share your project file and a screenshot of task manager Processes page showing the problem?

Here are the 2 projects I tested this on:
[Projects.zip](https://github.com/prusa3d/PrusaSlicer/files/13925742/Projects.zip)

Gryfang, 2024-01-14:
Sure. Here's a model that I just waited with supports off and support on
build plate only for way over twice the amount of time. I tested fuzzy skin
off, and supports generated almost instantly. With fuzzy skin on, the fuzzy
skin generated quickly, but the supports took way longer than expected.
I was using .05 / .05 on the fuzzy skin settings.

Gryfang, 2024-01-14:
This was also using organic supports. But like I said, it almost appeared
to be frozen longer than it should have. I opened up task manager, and it
was using one core. I use a laptop, but it's no slouch. 8c 16th Ryzen 9
5900HX, 32GB RAM, RTX 3070 GPU. Don't know if you are using the GPU for
anything, but even the onboard GPU isn't horrible.

DanielDC88, 2024-01-14:
@non-bin Here is the file I used and a video of the slicing process You will see the memory increase from the 1GB you mentioned from the 4m 9s mark onwards.
Video: [Prusaslicer Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/13932563/Prusaslicer.Issue.zip)
Project: [prusaslicer_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/13932560/prusaslicer_issue.zip)

closed 2026-09-29 by prusaslicer-bot[bot]