prusa3d/PrusaSlicer #10941: Prusa 2.6.0 freezes when trying to change color

kkrcma, 2023-07-02:
### Description of the bug

Prusa Slicer 2.6.0 freezes when right clicking on colored slider segment.

### Project file & How to reproduce

[J holes.zip](https://github.com/prusa3d/PrusaSlicer/files/11931205/J.holes.zip)
Open file
Slice
Right click on green colored slider 
The slicer will lock up. The only way to unfreeze is to end the task in task manager

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 + win64

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S

wallace11, 2023-07-04:
Same problem also occurs on Linux

neophyl, 2023-07-04:
As a counterpoint it does work fine on Windows 10.  Guess that's another reason to keep refusing Microsofts "upgrade"

wallace11, 2023-07-04:
@neophyl 
I can reproduce this consistently using HueForge models such as [this one](https://www.printables.com/model/492624-man-in-a-spacesuit-graphic-art). Just trying to **assign a color** to the last layer causes the program to crash.

Jan-Soustruznik, 2023-07-27:
Hello, @kkrcma , 

thank you for your report. We will try to fix in some next release. 

issue: SPE-1828

have a nice day