prusa3d/PrusaSlicer #10944: Stuck on Resolving Address (Mac)

lnjustin, 2023-07-03:
### Description of the bug

When attempting to upload and print to octoprint, the status message gets stuck on 'resolving address'. This issue did not exist in beta 3 (the last version we used). 
[image]

### Project file & How to reproduce

[Echo Stand.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11939751/Echo.Stand.3mf.zip)

1. Slice
2. Upload and Print to Octoprint

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

Mac OS Monterey 12.6.2

### Printer model

Prusa Mini+

JollyRgrs, 2024-01-19:
I am not getting the same thing (intermittent) printing to my Klipper/Mainsail instance. I am printing from Windows. nslookup and ping from command prompt both resolve to the correct IP, instantly and without any issues can be pinged. I am using PrusssaSlicer 2.7.1

whittaker007, 2024-03-25:
Same issue here. Was previously working. Printer settings: hostname + API key. Click Test button, get successful connection message. But gcode upload and print is stuck RESOLVING ADDRESS.

whittaker007, 2024-04-05:
Just installed 2.7.3 and the issue appears to be fixed.

kerios69780, 2024-09-23:
On my Mac (just updated the version of Macos and was issue is still present). It seems to appear when the Mac wakes from sleep. 
To fix, I need to close prusa slicer and start it again, load the project and it works.... Until next sleep cycle

closed 2026-09-29 by prusaslicer-bot[bot]