prusa3d/PrusaSlicer #10945: Organic support printing in mid-air

M4nt1c0r3, 2023-07-03:
### Description of the bug

When printing branches of organic supports, starting point of splitting branches is in mid-air, and printed material only falls down.

### Project file & How to reproduce

[image]
[image]
[Armor.zip](https://github.com/prusa3d/PrusaSlicer/files/11940428/Armor.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Wokna 10

### Printer model

Creativity ELF Pro, Manta M5P mainboard

M4nt1c0r3, 2023-07-03:
[image]
[image]
[G-code files.zip](https://github.com/prusa3d/PrusaSlicer/files/11940564/G-code.files.zip)

StephS, 2023-07-08:
I'm seeing the exact same issue. Auto generated supports.
Attached 3MF
[Bottle_cap_thrower.zip](https://github.com/prusa3d/PrusaSlicer/files/11986962/Bottle_cap_thrower.zip)

vazmuten, 2023-07-08:
Same issue using Organic Support in Prusa Mini.

[OrganicSupportMidAir.zip](https://github.com/prusa3d/PrusaSlicer/files/11992470/OrganicSupportMidAir.zip)

kubispe1, 2023-07-10:
We have it on our list, but it is still not resolved. Closing as a known issue.

closed 2023-07-10 by kubispe1