prusa3d/PrusaSlicer #10947: I have problems with my PrusaSlicer

MochuliChu, 2023-07-04:
### Description of the bug

My Prusa no longer shows me how much filament a print uses in meters. It also doesn't let me see the organic supports.

### Project file & How to reproduce

[mate stitch cuerpo entero.zip](https://github.com/prusa3d/PrusaSlicer/files/11952020/mate.stitch.cuerpo.entero.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Ender 3v2 pro

neophyl, 2023-07-04:
You have 2 extruders configured.  They are extruder 0 and extruder 1, but you have extruder number set to 2 for all extrusion types in Print Settings>Multiple Extruders.  That's why there is no extrusion in meters listed.  Neither extruder 0 or 1 are being used so there's all your gcode moves and extrusion values output but for T2 (Extruder 2) which you don't have according to your configuration.  

Same with the support, it briefly shows it then removes it as theres no actual support being produced with any of your configured extruders.  You should set the extruders to the default values of 0 and 1 and that way you will get the correct output.

MochuliChu, 2023-07-04:
Hello, thank you very much for your help. The issue of how much filament it uses was fixed when I switched to 'extruder 1'. But the support problem still occurs.

neophyl, 2023-07-04:
The default extruder according to the tooltips is 0 for support, if you change those to that then it appears. Not sure why this is required but it does solve your immediate problem.

MochuliChu, 2023-07-05:
Omg that solved the problem. Thanks a lot. <3

closed 2023-07-05 by MochuliChu