prusa3d/PrusaSlicer #10950: SegFault when using command line on MacOS

mkoistinen, 2023-07-04:
### Description of the bug

On MacOS 13.4.1 (Intel) attempting to get the basic help with:

`> /Applications/PrusaSlicer\ 2.6.0.app/Contents/MacOS/PrusaSlicer --help-fff`

It dumps out the help, then set faults (attached herein)

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [22670]
Path:                  /Applications/PrusaSlicer 2.6.0.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0+MacOS-x64-202306191415 (PrusaSlicer-2.6.0+MacOS-x64-202306191415)
Code Type:             X86-64 (Native)
Parent Process:        zsh [21476]
Responsible:           Electron [20265]
User ID:               501

Date/Time:             2023-07-04 17:38:14.3025 -0400
OS Version:            macOS 13.4.1 (22F82)
Report Version:        12
Bridge OS Version:     7.5 (20P5060)
Anonymous UUID:        384951CF-FA30-879B-0696-79795BFF6ED4

Sleep/Wake UUID:       911DA82D-86AE-49C4-BA34-6B72E6F4A36F

Time Awake Since Boot: 490000 seconds
Time Since Wake:       293542 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [22670]

VM Region Info: 0 is not in any region.  Bytes before following region: 4451065856
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1094de000-10caf2000    [ 54.1M] r-x/r-x SM=COW  ...S/PrusaSlicer

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x10957086f Slic3r::ConfigDef::print_cli_help(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, std::__1::function<bool (Slic3r::ConfigOptionDef const&)>) const + 1775
1   PrusaSlicer                   	       0x1094ef85c Slic3r::CLI::print_help(bool, Slic3r::PrinterTechnology) const + 2188
2   PrusaSlicer                   	       0x1094eb0b5 Slic3r::CLI::run(int, char**) + 19941
3   PrusaSlicer                   	       0x1094f04d0 main + 352
4   dyld                          	    0x7ff8045eb41f start + 1903

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fc40c7135b0  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x00007ff7b6a1c151  rsi: 0x0000000000000000  rbp: 0x00007ff7b6a1c1f0  rsp: 0x00007ff7b6a1bea0
   r8: 0x0000000000000002   r9: 0x0000000000000000  r10: 0x00000000ffffff00  r11: 0x00007ff6aa05f3e7
  r12: 0x00007ff7b6a1c180  r13: 0x00007ff7b6a1beb8  r14: 0x00007ff7b6a1bec0  r15: 0x00007fc40c7135e8
  rip: 0x000000010957086f  rfl: 0x0000000000010246  cr2: 0x0000000000000000
  
Logical CPU:     12
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

Thread 0 instruction stream:
  04 d4 02 48 8b 5d a8 0f-b6 83 e8 00 00 00 a8 01  ...H.]..........
[... 207 lines cut ...]
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
```

### Project file & How to reproduce

`/Applications/PrusaSlicer\ 2.6.0.app/Contents/MacOS/PrusaSlicer --help-fff` should do it.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0+MacOS-x64-202306191415

### Operating system

MacOS 13.4.1

### Printer model

n/a

mkoistinen, 2023-07-05:
I was also able to reproduce the same seg faults on an arm64 (M1) Mac running the same OS.

isotropo, 2023-08-03:
I was also able to reproduce this on a 2016 MacBook Pro Monterey Core i7. I have also had trouble with some transform options either doing nothing or resulting in a seg fault.

isotropo, 2023-08-03:
> I was also able to reproduce this on a 2016 MacBook Pro Monterey Core i7. I have also had trouble with some transform options either doing nothing or resulting in a seg fault.

Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [78004]
Path:                  /Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0+MacOS-x64-202306191415 (PrusaSlicer-2.6.0+MacOS-x64-202306191415)
Code Type:             X86-64 (Native)
Parent Process:        zsh [75539]
Responsible:           Terminal [75532]
User ID:               501

Date/Time:             2023-08-03 13:15:43.7564 -0700
OS Version:            macOS 12.6.8 (21G725)
Report Version:        12
Bridge OS Version:     3.0 (14Y910)
Anonymous UUID:        D698C064-2F19-CA58-FED2-87A668A36BB6

Sleep/Wake UUID:       47BC7BA1-1B41-4C36-B3B5-86E3EF10F327

Time Awake Since Boot: 80000 seconds
Time Since Wake:       1548 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [78004]

VM Region Info: 0 is not in any region.  Bytes before following region: 4531630080
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10e1b3000-1117c7000    [ 54.1M] r-x/r-x SM=COW  ...S/PrusaSlicer

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x10e24586f Slic3r::ConfigDef::print_cli_help(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool, std::__1::function<bool (Slic3r::ConfigOptionDef const&)>) const + 1775
1   PrusaSlicer                   	       0x10e1c485c Slic3r::CLI::print_help(bool, Slic3r::PrinterTechnology) const + 2188
2   PrusaSlicer                   	       0x10e1c00b5 Slic3r::CLI::run(int, char**) + 19941
3   PrusaSlicer                   	       0x10e1c54d0 main + 352
4   dyld                          	       0x1227dd52e start + 462

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fe7e58069a0  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x00007ff7b1d478f1  rsi: 0x0000000000000000  rbp: 0x00007ff7b1d47990  rsp: 0x00007ff7b1d47640
   r8: 0x0000000000000002   r9: 0x0000000000000000  r10: 0x00000000ffffff00  r11: 0x00007ff6a06b5b87
  r12: 0x00007ff7b1d47920  r13: 0x00007ff7b1d47658  r14: 0x00007ff7b1d47660  r15: 0x00007fe7e58069d8
  rip: 0x000000010e24586f  rfl: 0x0000000000010246  cr2: 0x0000000000000000
  
Logical CPU:     3
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

Thread 0 instruction stream:
  04 d4 02 48 8b 5d a8 0f-b6 83 e8 00 00 00 a8 01  ...H.]..........
  75 3b 48 89 c1 48 d1 e9-48 85 c9 74 3c 0f 57 c0  u;H..H..H..t<.W.
  0f 29 85 e0 fd ff ff 48-c7 85 f0 fd ff ff 00 00  .).....H........
  00 00 a8 01 0f 84 ab 00-00 00 4c 8b ab f0 00 00  ..........L.....
  00 4c 8b b3 f8 00 00 00-e9 a8 00 00 00 48 8b 8b  .L...........H..
  f0 00 00 00 48 85 c9 75-c4 48 8b b3 88 01 00 00  ....H..u.H......
 [48]8b 06 48 3b 46 08 0f-84 a6 01 00 00 4c 89 ef  H..H;F.......L..	<==
  48 8d 15 02 cf 3a 03 e8-05 0f 00 00 4c 89 ef 48  H....:......L..H
[... 134 lines cut ...]
    "queue" : "com.apple.main-thread"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "6297d96be2c9387df974efa4",
      "factorPackIds" : {

      },
      "deploymentId" : 240000008
    },
    {
      "rolloutId" : "6112e14f37f5d11121dcd519",
      "factorPackIds" : {
        "SIRI_TEXT_TO_SPEECH" : "634710168e8be655c1316aaa"
      },
      "deploymentId" : 240000231
    }
  ],
  "experiments" : [

  ]
}
}

mkoistinen, 2023-08-16:
Is there anything I can supply that will help this ticket see some activity? I have more than 50 models that I need to re-slice by hand each time a new IS/PA configuration is released.

mkoistinen, 2023-08-28:
Confirmed that the issue still exists in 2.6.1 RC1.

mkoistinen, 2023-09-10:
Looks like this was fixed by #11060.

closed 2023-09-10 by mkoistinen